pololu/vl53l0x-arduino #9: Adafruit library works but this library doesn't works correctly.

Kagyroy, 2017-07-11:
I tried this library but this library's "continuous.ino" outputs "65535" continuously..
I think the hardware connection is fine. 
So do I need some correction in my sample code.

tom-it, 2017-07-12:
Please recheck your wiring, 65535 is the highest available 16-bit number, so it looks like the measurement is timing out. Can you check if Single.ino works?

Kagyroy, 2017-07-26:
Thank you for answer my question.
Single.ino also outputs 65535.

richardwesthaver, 2017-10-17:
signed in to confirm this issue here as well, maybe something with the Adafruit boards? idk

GillHawk, 2017-10-20:
I had same issue (with 65535) and after opening and closing the SerialMonitor it started showing distance. But it freezes after few cycles if range is too high. In this state is unusable.

kevin-pololu, 2017-10-21:
For those who have used Adafruit's library successfully but are having trouble with ours (@Kagyroy and @richardwesthaver), are you using the Adafruit VL53L0X board, the Pololu board, or something else? What type of Arduino are you using?

@GillHawk, are you in the same situation where the Adafruit library works for you but not this one? If so, I'd like to know about your setup as well.

GillHawk, 2017-10-21:
Thank you for fast response. After some more tests, it seems that the library was OK. The problem was from some wired induction when I held the sensor cables in my hand (not actually touching any conducting parts..) when freezing occurred. When I avoided to touch the cables, it worked correctly.

mike-u, 2017-11-12:
I'd like to report that I'm also having this issue on an Adafruit sensor with an Arduino Pro Mini 3.3V which works with other sensors. I even added a   Serial.println('hello'); below the Serial.begin() and it outputs '27759' instead.

Kagyroy, 2017-11-16:
@kevin-pololu 
I used Uno & Nano.

Jongkeun, 2018-02-06:
@GillHawk . Are you sure?
I tested by using this library.
[image]

But Although I didn't touch cable, The number 8190 continues to appear.
Did you fix this issue?

Jongkeun, 2018-02-06:
I wanna use this library if it is fixed this issue.
This library has good document and API.
Please let me know how to fix it.

dufi2profor, 2018-05-26:
HI, continuous works fine on ESP8266. I even have two sensors wired to ESP8266 but that includes additional wires to XSHUT pins. I will post my sketch soon.

kevin-pololu, 2018-07-12:
@MauroMombelli Both our (Pololu's) and Adafruit's VL53L0X boards include pull-up resistors, so adding external resistors should not be necessary. That could be the issue if someone is trying to use this library with a different board that doesn't include the pull-ups, though.

zhivko, 2019-05-06:
I can confirm adafruit lib works with same sensor whereas this pololu lib doesn't work. I use it with .setAddres(0x29) since in adafruit lib 0x29 is default i2c address for this sensor.

I must note that I hadn't change wiring any kind just changed lib.

markg85, 2019-09-25:
Sooo.... That was an interesting fix...
I too had this issue and just started looking into what might be wonky.
Turn out that in this library Wire.begin() is never called. In the adafruit library it is part of the setup. The samples in this library also don't add a Wire.begin() call so that means that the default just don't work. Or well, in my case that is.

**So, again. Check if you actually call Wire.begin() before any data is written to the sensor!**

I do think this library should fix this somehow. That probably just involves editing the examples.

kevin-pololu, 2019-09-26:
@markg85 The example programs do call `Wire.begin()`; see [here](https://github.com/pololu/vl53l0x-arduino/blob/master/examples/Single/Single.ino#L37) and [here](https://github.com/pololu/vl53l0x-arduino/blob/master/examples/Continuous/Continuous.ino#L15).

If that is not what you mean, could you please clarify what you think the problem is?

markg85, 2019-09-26:
> @markg85 The example programs do call `Wire.begin()`; see [here](https://github.com/pololu/vl53l0x-arduino/blob/master/examples/Single/Single.ino#L37) and [...]

Hmm, don't know where i got that from. I opened an example and didn't see it. But i can't find it anymore.

Regardless, if people use examples from adafruit with this chip, they don't get a `Wire.begin()` so they might very easily forget to add it and get this very problem this topic is about. I too forgot to add it as i was modifying my code from the adafruit library to this one. Running an example is easy (just copy/paste) but changing existing code from one library to another requires... more careful reading sometimes ;) haha

I've got it working here just fine now with 2 sensors. It's an awesome library you folks have!

edugardo1, 2019-11-10:
hola, tengo el problema que todo funciona bien, pero cuando apunta al infinito me da un error y no lo puedo solucionar, ya que el sensor debe apuntar a un espacio abierto y al cortarse la distancia se activa.

markg85, 2019-11-10:
> hola, tengo el problema que todo funciona bien, pero cuando apunta al infinito me da un error y no lo puedo solucionar, ya que el sensor debe apuntar a un espacio abierto y al cortarse la distancia se activa.

Now again and in **English**, please.
I think it's clear from the other replies in here that English is the form of communication (in this topic at least) so why you even bother to respond in Spanish (and then either expect us to know Spanish or put your message through a translate service) is beyond me.

edugardo1, 2019-11-10:
OKAY. I'm sorry.
I have the problem that everything works fine, but when it points to an open place it gives an error and I can't solve it.
The sensor must point to an open space and when cutting the distance is activated.
On the other hand, he takes readings when he shouldn't do it, he has great instability.

kevin-pololu, 2019-11-14:
Hi, edugardo1.

It's not clear to me that you are having the same problem as the other people in this discussion, or that it is a problem with the library at all, so please post on the [Pololu forum](https://forum.pololu.com/) since that is a better place for troubleshooting. When you post, it would be helpful if you could describe what you are doing with the sensor, what behavior you are expecting, and what you are observing instead (including exactly what kind of error you are seeing, with a copy or screenshot of the entire error message if applicable).

edugardo1, 2019-11-15:
Hi @ Kevin-Pololu, I describe the project:
I work with a vl53lox, to take electronic times. This is done on a running track, when performing the test in a closed place everything works perfect, when I take it outdoors, it activates itself.
If I monitor it, it gives me 8191 or 8190, I have managed to block those instances but the problem persists, undoubtedly the intense light harms it.
How could I avoid that?
I apologize if I don't explain myself correctly, since I'm just an amateur.
thank you.

ninthclowd, 2020-03-31:
Not sure if this will help anyone else, but I stumbled across this thread trying to get this library to work with my adafruit style knockoff board. I couldn't get the thing to init() successfully.  In the end I found I needed to init with the 2v8 flag: 
`sensor.init(true);`

Good library :)

kevin-pololu, 2020-08-06:
Hi, ninthclowd.

The `io_2v8` argument to `init()` defaults to true if it is not specified, so you should not need to specify it explicitly. Please let us know if that is not the case for you and you would like to troubleshoot further.

Kevin

ryantm, 2020-08-06:
Hi, all.

It sounds like people have variously got their setups working without modifications to the library, so I'm going to close this. Feel free to post more details about your setup if you don't think this is the case.

Sincerely,
Ryan Mulligan

closed 2020-08-06 by ryantm