pololu/vl53l0x-arduino #14: Calibration

mochetts, 2017-10-11:
First of all, thanks for sharing your work and sorry for posting a question as an issue. Couldn't find any other better place to post it.

I now the readme says that this library is not providing a calibration API. However, I'd like to ask if it's possible to hardcode a calibration somewhere?

I really want to use this library as it's super thin and it works well with my setup, but I'm using the sensor under a cover glass making it to throw a lot of false positives.

Thanks in advance and sorry for the inconvenience.

kevin-pololu, 2017-10-21:
Unfortunately, the library doesn't provide a way to set cover glass calibration values and we don't have any plans to implement this feature for now. However, we'll keep this issue open in case we re-evaluate it later on, as it would be a nice feature to add.

satishbhandare, 2019-01-31:
I have been using this library, and am very happy with it.   

I want to cover my sensor with a glass cover, but the reading are incorrect. Hence it would be very helpful if you can add a routine for calibrating with a glass cover.

Many thanks in advance.

smenor, 2022-01-12:
Just adding a +1 on how useful it'd be to get the calibration here (and a -1000 to ST for having unbelievably lousy documentation which is making doing that myself at the very least needlessly frustrating and time consuming)

generosocarbone, 2022-01-14:
> Just adding a +1 on how useful it'd be to get the calibration here (and a -1000 to ST for having unbelievably lousy documentation which is making doing that myself at the very least needlessly frustrating and time consuming)

same here