pololu/vl53l0x-arduino #17: Possible wrong register value?

ssilverman, 2018-02-02:
https://github.com/pololu/vl53l0x-arduino/blob/22d997afc59625d14c3940d54dfd6d1800efa627/VL53L0X.cpp#L900

The ST source code uses 0x90 here, yours uses 0x92.

kevin-pololu, 2018-02-06:
Thanks, that does look like a mistake in our library. Have you tested it both ways and noticed any difference in performance?

ssilverman, 2018-02-08:
I haven't tested either case yet, but I'm not sure how the current SPAD count or aperture type can be read from a register that isn't the same as what the ST reference code uses. Unsure of the impact.

ssilverman, 2018-02-13:
I now know why it was 0x92. That's because I didn't notice the 32-bit read, so it turns out your code is correct. I went and reviewed it again.

closed 2018-02-13 by ssilverman