pololu/vl53l0x-arduino #18: Possible missing line that sets SYSTEM_INTERRUPT_CLEAR to 0x00?

ssilverman, 2018-02-13:
https://github.com/pololu/vl53l0x-arduino/blob/22d997afc59625d14c3940d54dfd6d1800efa627/VL53L0X.cpp#L1031

VL53L0X_ClearInterruptMask sets SYSTEM_INTERRUPT_CLEAR to 0x00 after setting it to 0x01. Curious why this isn't done here?

kevin-pololu, 2018-02-20:
I'm not so sure about my original reasoning on this one, but I think I found that only writing a 1 was sufficient to clear the interrupt and left out writing a 0 in the interest of simplifying the code (as well as not checking to make sure the interrupt is cleared).

For comparison, the datasheet for the similar VL6180X states for the SYSTEM__INTERRUPT_CLEAR register that "Writing a 1 to each bit will clear the intended interrupt", and we do the same thing (only writing a 1) in our VL6180X library.

kevin-pololu, 2018-05-30:
I am closing this issue as it doesn't sound like it's a problem, but please let me know if you notice any actual problematic behavior related to this.

closed 2018-05-30 by kevin-pololu