pololu/vl53l0x-arduino #26: How to use this library with digispark attiny85 board

PaulKovalov, 2018-09-02:
The problem is that this library doesn't work properly with attiny85, is it actually possible to use this library with attiny?

kevin-pololu, 2018-09-04:
Hi,

We have only written and tested this library for some of the most common standard Arduino boards, so it might not be practical to make it to work with all Arduino-compatible microcontrollers, but we can try to see if there is something simple preventing it from working with the ATTiny85. Can you provide more details about how it is not working? What behavior are you expecting and what do you see instead? Are you getting any error messages?

PaulKovalov, 2018-09-05:
I'm sorry, the problem was in power supply XD It works just excellent with your library, thank you for doing it!

closed 2018-09-05 by PaulKovalov

NicoBM10, 2020-12-22:
Hello, I'm trying to connect the VL53L0X to ATtiny85. Could you put the code here, please?