pololu/vl53l0x-arduino #39: Sensors on different I2C channels

TorbenSchreiter, 2020-03-12:
Hi there,

for an ESP32-based project, we needed a way to distribute sensors on the two separate hardware wires (`Wire.` and `Wire1.`). The modified version of the lib is attached ([VL53L0X-2xI2C.zip](https://github.com/pololu/vl53l0x-arduino/files/4323711/VL53L0X-2xI2C.zip)).

@kevin-pololu Changes are straight forward. Would be great if this could be integrated into the master branch at some point.

How to use:

```
#include <Wire.h>
#include "VL53L0X-2xI2C.h"

VL53L0X sensor;
VL53L0X sensor2;

Wire.begin(GPIOSDA,GPIOSCL); // initialize I2C w/ I2C pins from config
present = sensor.init(true, &Wire);

Wire1.begin(GPIOSDA2,GPIOSCL2); // initialize I2C w/ I2C pins from config
present2 = sensor2.init(true, &Wire1);
```

Cheers,
Torben

KurtE, 2020-08-01:
For what it is worth, I made a similar change, the main difference was I put the option Wire object on the constructor instead of on the init.  Either is fine.  

I went ahead and made a Pull Request (#45)  to see if they are receptive to this.  

There are some other features I am going to play with as well.  Trying to decide to do it here or with the Adafruit_VL5310X library, which supports this, and is made up the the ST SDK, and mainly only has a single Range function exported as part of the main .h/cpp files.

kevin-pololu, 2020-09-24:
Hi, Torben.

We just merged KurtE's pull request, although we ended up changing it so selecting a different I2C bus is done with separate functions (`setBus()` and `getBus()`) instead of through the constructor or `init()`. Thanks for your contribution though!

Kevin

closed 2020-09-24 by kevin-pololu