pololu/vl53l0x-arduino #40: Making rover move with VL53L0X attached on servo motor (to detect obstacle), IR Sensor to detect line/ path

Nisha-elsa, 2020-03-29:
Hi, I am trying to code a rover that will detect obstacle using vl53l0x and stay clear from it. Also, in case obstacle is not there, it should follow it's path which can be detected using two IR line sensors. 

I have written partial code so far. The rover works only with IR sensors in the previous version of code. When i try to incorporate the time of flight sensor VL53L0X too, the motors doesn't move. (Only the servo motor moves the 45 degrees as i have coded here. 

Here is my code block: 

#include<Wire.h>
#include<VL53L0X.h>
#include<Servo.h>

int servoPin =4;
VL53L0X sensor;
Servo Servo1;

#define HIGH_SPEED

int sensorval;
int obstacle =150;

const int NORMAl_SPEED  = 90; 
const int TURN_SPEED_H  = 100;

//Motor pins
const int motorSpeedPinA  = 9; 
const int motorDirPinA  = 7;  
const int motorSpeedPinB  = 10; 
const int motorDirPinB  = 8; 
const int modePin  = 11;

//IR pins
const int leftSensor  = A0; 
const int rightSensor  = A1; 

boolean leftFlag = 0;
boolean rightFlag = 0;

void setup() 
{
  pinMode(motorSpeedPinA, OUTPUT);
  pinMode(motorSpeedPinB, OUTPUT);
  pinMode(motorDirPinA, OUTPUT);
  pinMode(motorDirPinB, OUTPUT);
  pinMode(modePin, OUTPUT);
  digitalWrite(modePin, HIGH);
  pinMode(leftSensor, INPUT);
  pinMode(rightSensor, INPUT);
  Servo1.attach(servoPin);
  Servo1.write(90);
  Serial.begin(9600);
  Wire.begin();

  sensor.setTimeout(500);
  if (!sensor.init())
  {
    Serial.println("Failed to detect and initialize sensor!");
    while (1) {}
  }

  #if defined HIGH_SPEED
  // reduce timing budget to 20 ms (default is about 33 ms)
  sensor.setMeasurementTimingBudget(20000);
  #endif
  
  stop_vehicle();
}

void loop() 
{
  leftFlag = !digitalRead(leftSensor);
  rightFlag = !digitalRead(rightSensor);
  sensorval = sensor.readRangeSingleMillimeters();
  Serial.print(sensorval);
  if (sensor.timeoutOccurred()) { Serial.print(" TIMEOUT"); }
  Serial.println();
  
 if(sensorval < 150)
 {
 Servo1.write(45);
 delay(500);
 turn_left_vehicle(); 
 }
 if(leftFlag)
  {
    //Serial.println("Left");
    turn_right_vehicle();
  }
  else if(rightFlag) 
  {
    //Serial.println("Right");
    turn_left_vehicle(); 
  }
  else
  {
    forward_vehicle();
   
  }
  
  delay(20);
}

void stop_vehicle()
{
  analogWrite(motorSpeedPinA, 0);
  analogWrite(motorSpeedPinB, 0);  
}

void forward_vehicle()
{
  digitalWrite(motorDirPinA, HIGH);
  digitalWrite(motorDirPinB, HIGH);
  analogWrite(motorSpeedPinA, NORMAl_SPEED);
  analogWrite(motorSpeedPinB, NORMAl_SPEED);    
  
}

void turn_left_vehicle()
{
  digitalWrite(motorDirPinA, LOW);
  digitalWrite(motorDirPinB, HIGH);
  analogWrite(motorSpeedPinA, TURN_SPEED_H);
  analogWrite(motorSpeedPinB, NORMAl_SPEED);    
}

void turn_right_vehicle()
{
  digitalWrite(motorDirPinA, HIGH);
  digitalWrite(motorDirPinB, LOW);
  analogWrite(motorSpeedPinA, NORMAl_SPEED);
  analogWrite(motorSpeedPinB, TURN_SPEED_H);    
}

/*void object_detected()
{
 
    Servo1.write(45);
    delay(500);
    turn_left_vehicle();   
}*/

Nisha-elsa, 2020-03-29:
@kevin-pololu would you be able to suggest how to debug the code blocks?

kevin-pololu, 2020-08-07:
Hi,

If you are still looking for help getting this working, please post on the [Pololu Forum](https://forum.pololu.com/) (the GitHub issues are intended to track development of and problems with the library itself).

Kevin

closed 2020-08-07 by kevin-pololu