pololu/vl53l0x-arduino #52: VL53L0X and ESP8226 - Failed to detect ...

JanChrisKlein, 2021-07-22:
Hi,

I get a "Failed to detect and initialize sensor!" when using the Pololu VL53L0X Arduino library.
Arduino version ist 1.8.13. VL53L0X version is 1.3, downgrading to 1.2 didn't help. I'm compiling
for a Wemos D1 mini (ESP8226).

Using Your original code example 'single' will produce the above error and also cause repetitive reboots.
This can easily be remedied by inserting delays:

`  if (!sensor.init())
  {
    Serial.println("Failed to detect and initialize sensor!");
    while (1) {delay(1);}
  }`

instead of

`  if (!sensor.init())
  {
    Serial.println("Failed to detect and initialize sensor!");
    while (1) {}
  }`

Any suggestions?
Thanks,
Christoph

BlackWarriorNT, 2021-07-25:
I have exactly the same problem and I do not know how to solve it. it's very funny that if you just connect the sensor , then everything can work correctly

kevin-pololu, 2021-07-29:
Hello,

Could both of you post a question on the [Pololu forum](https://forum.pololu.com/) instead? That is a better place for troubleshooting, and it would be helpful if you could post some pictures of your system clearly showing all of your connections so we can check that they are okay.

The `while` loop is intended to basically stop the program if the sensor is not detected. As far as I know, it is an issue unique to the ESP8266 that an empty infinite loop will cause it to time out and reset, but it should be a simple fix to add a delay inside the loop to prevent that (like you said), so we'll try to fix that next time we update the library.

Kevin

closed 2021-07-31 by ryantm